DieBieEngineering/DieBieMS #11: Diode D5? One or two shunts?

BluePygmyOwl, 2019-03-17:
Hi Danny,
Big comliments on this. It looks really nice, and I thought I might give it a try to build this myself.

When I looked at the schematic, I had some questions.
- When a charger is plugged in, Chargedetect is pulled low. The voltage difference between charger and battery will turn on the optocoupler D2. First I thought D5 was some Zener Voltage Regulator diode, but then I saw that it isn't a zener. Why is D5 there then?

- Also, in the schematic you have R11 and R64 as shunt resistors (1mOhm), but on the pictures I only see one shunt placed on your boards. R11 is rated for 5W. Isn't that short when using proper discharge loads (electric board)? If R64 is added, does this imply a software setting that has to be changed?
[image]

Thanks in advance

DieBieEngineering, 2019-03-18:
The two shunts are there to allow for different component population options ( to experiment ). In the Altium project you can select population variants, with every variant I made the I always have only the top shunt mounted. 

I can totally imagine how strange it looks! Some LEDs can handle only a limited voltage in the reverse direction, this diode prevents that voltage to be created. This circuit also changes in a evaluation branch ( HW V0.10 ), there I also implement a P fet as ideal input diode.

BluePygmyOwl, 2019-03-18:
Hi Danny,
Thanks. Should I wait with my build for V0.10? Or does that take some significant time before you can publish? I am kind of amped to start to build this, but I do like the idea of the ideal input diode. Will you use a comparator instead of the opto for ChargeDetect?

DieBieEngineering, 2019-03-18:
It is already in GitHub on a branch, I will look it up and let you know. It
is also tested and works!