DieBieEngineering/DieBieMS #13: How turn on this BMS?

rahmaevao, 2019-03-18:
Hi. I assembled the device.
HW 0.8
I connected battery. I upload firmware. Device connecting in DieBieMSTool.
I connect pin J14:2 to GND. Terminal DieBieMS says the button is pressed.

> -----Battery Pack Status-----
Pack voltage          : 24.52V
Pack current          : 0.04A
Low  current          : 0.04A
High current          : 0.00A
State of charge       : 0.0%
Remaining capacity    : 0.00Ah
Operational state     : Power down
Load voltage          : 0.00V
Cell voltage high     : 3.506V
Cell voltage low      : 3.503V
Cell voltage average  : 3.505V
Cell voltage mismatch : 0.003V
Discharge enabled     : False
Charge enabled        : False
Power button pressed  : True
---End Battery Pack Status---

But BMS not connected LOAD and battery. OpState remains Power down.

How turn on this BMS?

DieBieEngineering, 2019-03-18:
Ok ltc seems to work and ISL as well!

However the power button should not indicate pressed unless it is actually
pressed. You don't have a button connected right?

rahmaevao, 2019-03-19:
Thanks for the quick feedback.
I really like this BMS.
> Ok ltc seems to work and ISL as well!

Yes, it work.

> However the power button should not indicate pressed unless it is actually
> pressed. You don't have a button connected right?

I not have special button, but i connect pin 2 J14 connector to pin 1 J14 connector.
If i connect pin 2 to pin 1, terminal says "Power button pressed  : True".
If pin 2 is not connected to pin 1,  terminal says "Power button pressed  : False".

DieBieEngineering, 2019-03-19:
Ok in that case it works as it should. The power button input should be a
momentary switch.

I'll explain a bit more in a sec. Did you also find the skateboard forum?

rahmaevao, 2019-03-20:
Oh, Denny. Your forum not acessed in my region (Russia). Proxy in my provider not let throgh vpn.
I use cashe google for reading forum.((

rahmaevao, 2019-03-21:
Oh, my BQ76200PW destroyed.
I changed it and everything works.

closed 2019-03-21 by rahmaevao