DieBieEngineering/DieBieMS #15: Power cutting off. Vesc indicating low vaultage Fault

Wraith3333, 2019-07-27:
When running on the bench, everything works fine but the moment I step on the board, power is cut off to the vescs and the board dies. Checked all the wiring and updated to v.27. running 16Ah 10s6p Li-ion pack with DBMS

DieBieEngineering, 2019-07-27:
Do you have pictures of your setup? 

Can you video the BMS tool showing the real-time cell voltages?

Sounds like a wiring issue or a low voltage cutoff...

Wraith3333, 2019-07-29:
[image]

[image]

The wiring is solid, nothing loose. It cuts off randomly on the bench as well without any load. I can't upload videos but I've attached a screenshot of the Realtime Data

Wraith3333, 2019-08-02:
Switching on the board with the display of the bms connected shows random shut offs with the display showing "bye" the same way it would when manually shutting it down via a power switch

Wraith3333, 2019-08-02:
[image]
unsure why the values indicate fail when running testbms on the terminal. could this be the reason for the shut off