DieBieEngineering/DieBieMS #16: SAFETY Signal on GPIO2

DanieGieb, 2020-03-04:
Hello, 

first of all really nice project! I am currently making a special Version for a custom battery pack.
So thank you very much for sharing!

I have a question:
What is the purpose for connecting the SAFETY signal to GPIO2 of the LTC6803?
I guess its for the reason when something went wrong with communication due to a uC brownout for example?
But why did you use GPIO2 for that? I think the WDTB pin is a better choice for that?
Like the datasheet suggests on page 15:

> WATCHDOG TIMER CIRCUIT
The LTC6803 includes a watchdog timer circuit. The watchdog timer is on for all modes except CDC = 0. The watchdog timer times out if no valid command is received for 1 to 2.5 seconds. When the watchdog timer circuit times out, the WDTB open-drain output is asserted low and the configuration register bits are reset to their default (power-up) state. In the power-up state, CDC is 0, the S outputs are off and the IC is in the low power standby mode. The WDTB pin remains low until a valid command is received. The watchdog timer provides a means to turn off cell discharging should communications to the MPU be interrupted. There is no need for the watchdog timer at CDC = 0 since discharging is off. The open-drain WDTB output can be wire ORd with other external open-drain signals. Pulling the WDTB signal low will not initiate a watchdog event, but the CNFGO bit 7 will reflect the state of this signal. Therefore, the WDTB pin can be used to monitor external digital events if desired.

If using GPIO2 and the uC sets this pin to zero, the power stage can be switched on. When the uC gets stucked or something went wrong the pin stays at low level, so the power stage will remain enabled!?
This would not happen if SAFETY is connected to the WDTB pin, which uses the LTC6803 internal watchdog?

Or is there any special purposes why you used GPIO2?

Here is my current implementation of the WDTB watchdog feature:
[image]

This also has the additional advance that the ISL28022 can also interrupts discharge/charge if an error occurs. (Overcurrent for example).
I think this should be much faster than doing this with the uC. 
And once the ISL28022 is successfully configured and the software later stucks the overcurrent protection is still available.

Kind Regards
Danie

DieBieEngineering, 2020-03-06:
This is a system to disable the out and input when the watchdog triggers of the LTC (the LTC resets also its GPIO on WDT overflow). I use an GPIO on the LTC because over SPI I tell the LTC to pull the wire low (low being the charge / discharge are allowed to become high). So this is exactly as you have understood it operates.

The drawback of using the WDT is that it works the other way around, and you dont have a clear understanding of why the pin is asserted (any communication that is valid will reset the WDT if I recall correctly), where in the GPIO approach you have the uC deliberately ask the pin to go low.

( I actually first build it with the use of the WDT pin, but then moved to using the GPIO because of the added features described )

DanieGieb, 2020-03-06:
> This is a system to disable the out and input when the watchdog triggers of the LTC (the LTC resets also its GPIO on WDT overflow). I use an GPIO on the LTC because over SPI I tell the LTC to pull [...]

That clearly makes sense to me! You are right - any valid communication will reset the WDT!
Thank you for your explanation!
I will use GPIO2. Because i don't have space for the SD card i will connect the WDT pin to the uC. 
The ISL Interrupt signal is open drain, so i will connect it over 0R to the Safety signal.
So the ISL can also disable the charge and discharge FETs. 
I think this will add another good safety feature to the device, because after the ISL is successfully configured, the software can stuck and overcurrent over- and undervoltage cutoff is still possible by hardware.

Another short question:
Is there a UART left on the STM? I plan to implement a connector for a serial bluetooth interface module like the cheap HC05, because the SPI wireless module is not available in germany :/
Could the mobile app of the BMS tool use this bluetooth connection or is connection only possible with your implemented wireless module? 
I think this should work!?